laszlodaniel/SmartBatteryHack #7: a problem on define arrays in c++/arduino

youxiaojie, 2020-04-22:
dear friend, I am not sure whether you make mistake in define array.
`    uint16_t smbus_reg_dump_payload_length = 3*(smbus_reg_end - smbus_reg_start + 1) + 2;
    uint8_t smbus_reg_dump_payload[smbus_reg_dump_payload_length]; // max 770 bytes`

what you mean is define a dynamic-size array? Using variable as size when defining array  is not allowed in C, is it permit in arduino?

laszlodaniel, 2020-04-22:
It's not permitted in Arduino either.
I did this because after the function ends (leaves scope) this array is removed from memory and next time the function is called the array is created again. I like to first calculate the length in a separate variable and pass its value as the array length. Still it's a fixed value calculated once and it's not changing.

youxiaojie, 2020-04-22:
I got it. In C99 standard it is ok. My book is 1996,and I have not used C for decades :( I am outdated.

closed 2020-04-23 by youxiaojie