laszlodaniel/SmartBatteryHack #8: dear friend, what is 0x17? why if we get 0x17, we will wait?

youxiaojie, 2020-04-23:
counter = 0;
        while (counter < 50)
        {
            data = read_byte(PeekROMByte); // get memory value and save it to the buffer
            if (data != 0x17) break;
            counter++;
            delay(10);
        }

laszlodaniel, 2020-04-23:
It's an artifact of the Sanyo-firmware. During reading if error occurs 0x17 gets returned. In this while-loop I give 50 chances to the firmware to get things right. After 50 retry Arduino gives up and saves 0x17 as the real value.
Unfortunately there's no reading error flag in the I2C-protocol that we can query so I have to get messy like this.

youxiaojie, 2020-04-23:
I find some i2c program also showed ACK and NACK byte, do you know what value of it?

youxiaojie, 2020-04-23:
"It's an artifact of the Sanyo-firmware. During reading if error occurs 0x17 gets returned. In this while-loop I give 50 chance to the firmware to get things right. After 50 retry Arduino gives up and saves 0x17 as the real value."

for example, I try to dump data flash in sealed mode, the chip will return an error code, maybe 0x17?

youxiaojie, 2020-04-23:
I have changed reg_dump function, please take a look. is it ok?

`void smbus_reg_dump(void)
{
    uint16_t smbus_reg_dump_payload_length = 0;
    uint8_t smbus_reg_dump_payload[1018]; // payload max 1018 bytes each pacckage due to uart protocol limit
	smbus_reg_dump_payload[smbus_reg_dump_payload_length++] = smbus_reg_start; // start register
    smbus_reg_dump_payload[smbus_reg_dump_payload_length++] = smbus_reg_end; // end register
    uint16_t data = 0;
    
    for (uint16_t i = smbus_reg_start; i < (smbus_reg_end + 1); i++)
    {
        switch(i)
		{
			case ManufacturerName: //0x20 string
			case DeviceName: //0x21 string
			case DeviceChemistry: //0x22 string
			case ManufacturerData: //0x23 hex data
			#ifdef (BQ20ZXX)
			case Authenticate: //0x2f string
            #endif
			{
			    uint8_t block_length = read_block(i, buffer, buffer_length); 
                smbus_reg_dump_payload[smbus_reg_dump_payload_length++] = i; // register first
				for(uint8_t j=0; j< block_length; j++)
				    smbus_reg_dump_payload[smbus_reg_dump_payload_length++] = dumped_block[j]; // dump block
                break;
			}
            #ifdef (BQ208X)
			case PackStatusPackConfig: //0x2f word
			#endif
			default:
		    data = read_word(i, reverse_read_word_byte_order);
            smbus_reg_dump_payload[smbus_reg_dump_payload_length++] = i; // register first
            smbus_reg_dump_payload[smbus_reg_dump_payload_length++] = (data >> 8) & 0xFF; // high byte of the word there
		    smbus_reg_dump_payload[smbus_reg_dump_payload_length++] = data & 0xFF; // low byte of the word there
		}
    }
    
    send_usb_packet(status, sd_smbus_reg_dump, smbus_reg_dump_payload, smbus_reg_dump_payload_length);
}`

just the standard sbs command.

youxiaojie, 2020-04-26:
[INFO] SMBus register dump details (20-23):
[20]: 45 43 53 // ManufacturerName: ECS
[21]: 33 32 30 43 31 32 // DeviceName: 320C12
[22]: 4C 49 4F 4E // DeviceChemisty: LION
[23]: 06 58 1B B8 67 01 00 11 00 C8 00 7B 2A // ManufacturerData: 06 58 1B B8 67 01 00 11 00 C8 00 7B 2A

laszlodaniel, 2020-04-27:
> I try to dump data flash in sealed mode, the chip will return an error code, maybe 0x17?

It's not that kind of error code. More like a timout on the fuel gauge IC's end. It doesn't matter how slowly I'm reading the SMBus, it just happens.

The new smbus_reg_dump function looks cool! Did you finish the register interpretation in C#? If so please make a pull request and I gladly merge it into the master branch.

youxiaojie, 2020-04-27:
not finished yet... polishing...

youxiaojie, 2020-05-02:
1.Adding chips detect
[RX->] Device settings
3D 00 0D 83 01 03 2B 5C 12 C0 08 24 01 33 2A 71 E8

[INFO] Word byte-order: reverse read/write

[INFO] Design voltage: 11.1V Design capacity: 4800mAH

[INFO] Chip: BQ2084 Firmware ver 1.33 Hardware ver 2a.71

[INFO] Word byte-order: reverse read/write

[INFO] Design voltage: 11.1V Design capacity: 4800mAH

[INFO] Chip: BQ2084 Firmware ver 1.33 Hardware ver 2a.71

2. debug interpretation. since I got the chipid, I will show interpretation according to each platform
[INFO] SMBus register dump details (40-7F):
--CAUTIOUS! Lookup datasheet for details of following Ext Reg. Value 0x1717 maybe an error when reading reg. Read it individually and dump reg 0x16 to view error code--
[40]: 17 17 // Ext_CMD: HEX_1717 BIN_00010111_00010111 DEC_5911 Or a 23-byte block. 
[41]: 17 17 // Ext_CMD: HEX_1717 BIN_00010111_00010111 DEC_5911 Or a 23-byte block. 
[42]: 00 00 // Ext_CMD: HEX_0000 BIN_00000000_00000000 DEC_0 Or a 0-byte block. 
[43]: 00 20 // Ext_CMD: HEX_0020 BIN_00000000_00100000 DEC_32 Or a 32-byte block. 
[44]: 09 06 // Ext_CMD: HEX_0906 BIN_00001001_00000110 DEC_2310 Or a 6-byte block. 
[45]: 32 3A // Ext_CMD: HEX_323a BIN_00110010_00111010 DEC_12858 Or a 58-byte block. BQ20ZXX: AFEData. Read 58-byte-block manually. BQ208X: VPack: 12858 mV
[46]: 00 0B // Ext_CMD: HEX_000b BIN_00000000_00001011 DEC_11 Or a 11-byte block. BQ20ZXX: FETControl: AFE GPOD Disabled. ZVCHG FET On. CHG FET Off. DSG FET On. 
[47]: 32 3A // Ext_CMD: HEX_323a BIN_00110010_00111010 DEC_12858 Or a 58-byte block. 
[48]: 32 3A // Ext_CMD: HEX_323a BIN_00110010_00111010 DEC_12858 Or a 58-byte block. 
[49]: 32 3A // Ext_CMD: HEX_323a BIN_00110010_00111010 DEC_12858 Or a 58-byte block. 
[4A]: 32 3A // Ext_CMD: HEX_323a BIN_00110010_00111010 DEC_12858 Or a 58-byte block. 
[4B]: 32 3A // Ext_CMD: HEX_323a BIN_00110010_00111010 DEC_12858 Or a 58-byte block. 
[4C]: 32 3A // Ext_CMD: HEX_323a BIN_00110010_00111010 DEC_12858 Or a 58-byte block. 
[4D]: 32 3A // Ext_CMD: HEX_323a BIN_00110010_00111010 DEC_12858 Or a 58-byte block. 
[4E]: 32 3A // Ext_CMD: HEX_323a BIN_00110010_00111010 DEC_12858 Or a 58-byte block. 
[4F]: 32 3A // Ext_CMD: HEX_323a BIN_00110010_00111010 DEC_12858 Or a 58-byte block. BQ20ZXX: StateOfHealth: 58% Deterioration Warning. 
[50]: 17 17 // Ext_CMD: HEX_1717 BIN_00010111_00010111 DEC_5911 Or a 23-byte block. BQ20ZXX: SafetyAlert: Charge OC. Tier-2 Charge OC. Pack UV. Pack OV. Host WD. Discharge OC. Charge short-circuit. Discharge short-circuit. BQ208X: Word, Writing a byte to dataflash (address ≤ 0xff).
[51]: 17 17 // Ext_CMD: HEX_1717 BIN_00010111_00010111 DEC_5911 Or a 23-byte block. BQ20ZXX: SafetyStatus: Charge OC. Tier-2 Charge OC. Pack UV. Pack OV. Host WD. Discharge OC. Charge short-circuit. Discharge short-circuit. BQ208X: Word,Setting the address (≤ 0xff) of the dataflash byte to read. 
[52]: 00 00 // Ext_CMD: HEX_0000 BIN_00000000_00000000 DEC_0 Or a 0-byte block. BQ20ZXX: PFAlert: BQ208X: Reading a byte from dataflash. 
[53]: 17 17 // Ext_CMD: HEX_1717 BIN_00010111_00010111 DEC_5911 Or a 23-byte block. BQ20ZXX: PFStatus: Open Thermistor. Charge Safty OC. Periodic AFE Com Fault. Permanent AFE Com Fault. Cell-Imbalance. Charge Safety OT. Safety OV. External PF Input. 
[54]: 32 3A // Ext_CMD: HEX_323a BIN_00110010_00111010 DEC_12858 Or a 58-byte block. BQ20ZXX: OperationStatus: FULL ACCESS Mode. SEALED Mode. DataFlash CheckSum Value Generated. Bit09. Discharge Fault. Discharge Disabled for Current Issue. Discharge Inhibited for High Temp. VOK for Qmax Update. 
[55]: 32 3A // Ext_CMD: HEX_323a BIN_00110010_00111010 DEC_12858 Or a 58-byte block. BQ20ZXX: ChargingStatus: Charging suspended. Precharging. Maintenance charging. Bit09. Discharge Fault. Bit04. Bit03. Overcharge fault. 
[56]: 74 12 // Ext_CMD: HEX_7412 BIN_01110100_00010010 DEC_29714 Or a 18-byte block. 
[57]: 32 3A // Ext_CMD: HEX_323a BIN_00110010_00111010 DEC_12858 Or a 58-byte block. BQ20ZXX: ResetData: Full Resets: 50 Partial Resets: 58
[58]: 00 40 // Ext_CMD: HEX_0040 BIN_00000000_01000000 DEC_64 Or a 64-byte block. BQ20ZXX: WDResetData: Watchdog resets :64timesBQ208X: 64bytes Reads/Writes data flash
[59]: 00 40 // Ext_CMD: HEX_0040 BIN_00000000_01000000 DEC_64 Or a 64-byte block. BQ208X: 64bytes Reads/Writes data flash
[5A]: BF 40 // Ext_CMD: HEX_bf40 BIN_10111111_01000000 DEC_48960 Or a 64-byte block. BQ208X: 64bytes Reads/Writes data flash
[5B]: AC 40 // Ext_CMD: HEX_ac40 BIN_10101100_01000000 DEC_44096 Or a 64-byte block. BQ208X: 64bytes Reads/Writes data flash
[5C]: 00 37 // Ext_CMD: HEX_0037 BIN_00000000_00110111 DEC_55 Or a 55-byte block. BQ2084: 55bytes Reads/Writes data flash
[5D]: 32 3A // Ext_CMD: HEX_323a BIN_00110010_00111010 DEC_12858 Or a 58-byte block. BQ20ZXX: AverageVoltage: 12858 mV
[5E]: 32 3A // Ext_CMD: HEX_323a BIN_00110010_00111010 DEC_12858 Or a 58-byte block. 
[5F]: 32 3A // Ext_CMD: HEX_323a BIN_00110010_00111010 DEC_12858 Or a 58-byte block. 
[60]: 00 10 // Ext_CMD: HEX_0010 BIN_00000000_00010000 DEC_16 Or a 16-byte block. BQ20ZXX: UnSealKey: Read 16-byte-block manually. BQ208X: 16 bytes Reads/Writes data flash
[61]: 43 10 // Ext_CMD: HEX_4310 BIN_01000011_00010000 DEC_17168 Or a 16-byte block. BQ20ZXX: FullAccessKey: Read 16-byte-block manually. BQ208X: 16 bytes Reads/Writes data flash
[62]: 32 10 // Ext_CMD: HEX_3210 BIN_00110010_00010000 DEC_12816 Or a 16-byte block. BQ20ZXX: PFKey: Read 16-byte-block manually. BQ208X: 16 bytes Reads/Writes data flash
[... cut ...]
[7C]: 32 3A // Ext_CMD: HEX_323a BIN_00110010_00111010 DEC_12858 Or a 58-byte block. BQ20ZXX: DataFlashSubClassPage5: Read 58-byte-block manually. 
[7D]: 32 3A // Ext_CMD: HEX_323a BIN_00110010_00111010 DEC_12858 Or a 58-byte block. BQ20ZXX: DataFlashSubClassPage6: Read 58-byte-block manually. 
[7E]: 32 3A // Ext_CMD: HEX_323a BIN_00110010_00111010 DEC_12858 Or a 58-byte block. BQ20ZXX: DataFlashSubClassPage7: Read 58-byte-block manually. 
[7F]: 32 3A // Ext_CMD: HEX_323a BIN_00110010_00111010 DEC_12858 Or a 58-byte block. BQ20ZXX: DataFlashSubClassPage8: Read 58-byte-block manually.

closed 2020-05-05 by youxiaojie