laszlodaniel/SmartBatteryHack #10: Can not write word

menjic, 2020-06-17:
If I for example read 0x17 I get 3 cycles, if I want to write this to be zero I think I need to write 00 00, but on next read I get 00 16, and what ever I try to do I van write just first byte. This is from GUI, when I click word to write word in upper window it sais for example 00 00 is data writen or sent and 2 bytes writen but when I read again it is always
 xx 16

laszlodaniel, 2020-06-17:
I can't reproduce the problem but have you tried changing the word byte-order?
What fuel gauge IC are you trying to communicate with?
Please paste the Communication textbox content here so I can see every step you take.

menjic, 2020-06-17:
I am trying to write to A2168 which is pretty much copy of BQ20Zxx, it may be Arduino board or win10 or 7 other things I looked into your code in visual studio I can not see anything wrong, I will set breakpoint before write and send you that or suggest me where else would be more useful.

Yes I did try to reverse write and read it is strange like always 00 becomes what ever I want and 16 stays 16.

Last night I tried 0x18 battery capacity, there was value 08 98 which is translated as 2200 mAH, I sent 08 99 just for test and got back 08 16

Here is code:

`[INFO] Connecting to COM4

[<-TX] Handshake request (COM4)
3D 00 02 01 00 03

[INFO] Device is not responding at COM4

[INFO] Connecting to COM4

[<-TX] Handshake request (COM4)
3D 00 02 01 00 03

[RX->] Handshake response
3D 00 08 81 00 53 42 48 41 43 4B 35

[INFO] Handshake OK: SBHACK

[INFO] Device connected (COM4)

[RX->] Device settings
3D 00 0B 83 01 03 1C E8 08 16 00 00 00 00 B4

[INFO] Word byte-order: reverse read/write

[INFO] Design voltage: 7.4 V Design capacity: 2070 mAH

[INFO] Chip: 0x0000 Firmware ver 0.0`

second part:
`[INFO] Reg.: 17
       Data: 16

[<-TX] Read word data
3D 00 03 04 02 17 20

[RX->] Word data received
3D 00 05 84 02 17 00 16 B8

[INFO] Reg.: 17
       Data: 00 16

[<-TX] SMBus register dump request
3D 00 04 02 03 17 19 39

[RX->] SMBus register dump (17-19)
3D 00 0D 82 03 17 19 17 00 16 18 08 16 19 1C E8 42

[INFO] SMBus register dump details (17-19):
[17]: 00 16 // CycleCount: 22 Cycles
[18]: 08 16 // DesignCapacity: 2070 mAh
[19]: 1C E8 // DesignVoltage: 7400 mV`

Third part:
`[<-TX] Read word data
3D 00 03 04 02 17 20

[RX->] Word data received
3D 00 05 84 02 17 00 16 B8

[INFO] Reg.: 17
       Data: 00 16

[<-TX] Write word data
3D 00 05 05 02 17 00 00 23

[RX->] Word data write response
3D 00 06 85 02 17 00 00 02 A6

[INFO] Reg.: 17
       Data: 00 00
       # of bytes written: 02

[<-TX] Read word data
3D 00 03 04 02 17 20

[RX->] Word data received
3D 00 05 84 02 17 00 16 B8

[INFO] Reg.: 17
       Data: 00 16

[<-TX] SMBus register dump request
3D 00 04 02 03 17 19 39

[RX->] SMBus register dump (17-19)
3D 00 0D 82 03 17 19 17 00 16 18 08 16 19 1C E8 42

[INFO] SMBus register dump details (17-19):
[17]: 00 16 // CycleCount: 22 Cycles
[18]: 08 16 // DesignCapacity: 2070 mAh
[19]: 1C E8 // DesignVoltage: 7400 mV

[<-TX] Read word data
3D 00 03 04 02 17 20

[RX->] Word data received
3D 00 05 84 02 17 00 16 B8

[INFO] Reg.: 17
       Data: 00 16

[<-TX] Write word data
3D 00 05 05 02 17 00 00 23

[RX->] Word data write response
3D 00 06 85 02 17 00 00 02 A6

[INFO] Reg.: 17
       Data: 00 00
       # of bytes written: 02

[<-TX] Read word data
3D 00 03 04 02 17 20

[RX->] Word data received
3D 00 05 84 02 17 00 16 B8

[INFO] Reg.: 17
       Data: 00 16`

80% I am doing something wrong but don't see it, 19% it is something in hardware, 1% it is something in code but my problem is I don't know where to even start troubleshooting so I started here in hope someone would recognize the problem and help me. Thank you.

laszlodaniel, 2020-06-17:
Oh I see what you want to do: you want to clear the cycle count. That won't work until you first unseal the battery controller. Now it's in sealed mode and you can't do anything that changes the memory. I'm not familiar with the unsealing this controller but it goes like writing the unseal key to the manufacturer access register. Look it up.

menjic, 2020-06-17:
Well I did think about that, but how I am able to write one part and not other, If I understand protection if it is in seal state write is forbiden it is not half forbiden. 

For example my cycle count was 2, now it is 22 I wrote that, my capacity was 2200 now it is 2070 ?

value of 0x18 was 08 98, I tried to write 08 99, now I read 0x18 and it is 08 16 so you see I can write first half of number is that how protection works ?

In documentation is stated if chip is in sealed state that it is read only that is what is strange to me how come I can write half of the word ?

menjic, 2020-06-17:
You can see in log I am sending "xx yy" and it writes only "xx" part of word ?

laszlodaniel, 2020-06-17:
Strange indeed. You should send the unseal key in two parts. There's a timeout in the controller that expects another message within seconds. You should be able to do it using the clipboard.

> You can see in log I am sending "xx yy" and it writes only "xx" part of word ?

I see both bytes are written.
Are your referring to this line?
`# of bytes written: 02`
This is just a counter, 2 bytes for a word write and n-bytes for block write.

Proper sealed mode shouldn't behave like this. Perhaps there's some difference in your A2168 after all.

menjic, 2020-06-17:
Yes but it is literaly using BQ firmware see this : 

[INFO] SMBus register dump details (20-30):
[20]: 59 58 58 4D // ManufacturerName: YXXM
[21]: 30 36 31 33 38 34 // DeviceName: 061384
[22]: 4C 49 4F 4E // DeviceChemisty: LION
[23]: 42 34 31 // ManufacturerData: 42 34 31
[24]: FF FF // Reserved: 65535
[25]: FF FF // Reserved: 65535
[26]: FF FF // Reserved: 65535
[27]: FF FF // Reserved: 65535
[28]: FF FF // Reserved: 65535
[29]: FF FF // Reserved: 65535
[2A]: FF FF // Reserved: 65535
[2B]: FF FF // Reserved: 65535
[2C]: FF FF // Reserved: 65535
[2D]: FF FF // Reserved: 65535
[2E]: 00 00 // Reserved: 0
[2F]: 00 14 // OptionalMfgFunction5: HEX_0014 BIN_00000000_00010100 DEC_20 Or a 20-byte block. BQ20ZXX: Authenticate. Read 20-byte-block manually. 
[30]: D4 0A // Reserved: 54282

[<-TX] SMBus register dump request
3D 00 04 02 03 30 40 79

[RX->] SMBus register dump (30-40)
3D 00 37 82 03 30 40 30 D4 0A 31 D4 0A 32 00 00 33 00 00 34 00 00 35 00 08 36 00 00 37 04 08 38 
04 08 39 04 08 3A 04 08 3B 0B C5 3C 00 00 3D 00 00 3E 0B E4 3F 0A 79 40 FF FF 18

[INFO] SMBus register dump details (30-40):
--CAUTIOUS! Lookup datasheet for details of Ext Reg. Value 0x1717 maybe an error when reading reg. Read it individually and dump reg 0x16 to view error code--
[30]: D4 0A // Reserved: 54282
[31]: D4 0A // Reserved: 54282
[32]: 00 00 // Reserved: 0
[33]: 00 00 // Reserved: 0
[34]: 00 00 // Reserved: 0
[35]: 00 08 // Reserved: 8
[36]: 00 00 // Reserved: 0
[37]: 04 08 // Reserved: 1032
[38]: 04 08 // Reserved: 1032
[39]: 04 08 // Reserved: 1032
[3A]: 04 08 // Reserved: 1032
[3B]: 0B C5 // Reserved: 3013
[3C]: 00 00 // OptionalMfgFunction4 (VCell4 for BQ): 0 mV
[3D]: 00 00 // OptionalMfgFunction3 (VCell3 for BQ): 0 mV
[3E]: 0B E4 // OptionalMfgFunction2 (VCell2 for BQ): 3044 mV
[3F]: 0A 79 // OptionalMfgFunction1 (VCell1 for BQ): 2681 mV
[40]: FF FF // Ext_CMD: HEX_ffff BIN_11111111_11111111 DEC_65535 Or a 255-byte block.

laszlodaniel, 2020-06-17:
I can't help you with this specific problem. This part of the code is a contribution from another member and is harder to understand than my original attempt at the BQ8050 Sanyo-firmware.

I only created an interface between computer and battery. Actual data transfer and behavior is different among controllers and is rarely discussed.

menjic, 2020-06-17:
Ok I will continue to experiment when I have time and will report back my findings, but that could take some time, should I close this and make new one when I find something or leave it opet if someone else finds this and has something to say.

And other question, probably I have things to add to your c# program do you want to include them or how would I continue with this, I am interested in improving and extending this project do I fork and continue or send you requests when I have something useful or new to add ?
It would be shame for this to stop by my opinion, what do you think ?

laszlodaniel, 2020-06-17:
Feel free to leave this issue open, no pressure.
Fork the repository, make your changes at your side and send a pull request to me and I'll merge the changes.
Thanks for taking your time for improvements!

menjic, 2020-06-17:
No thank you, you did all the work When and if I do close to your part than you can thank me :)

youxiaojie, 2020-07-23:
wait ,wait,what is problem? is there some bug?

youxiaojie, 2020-07-23:
I try to test with my bettery to confirm. the read write code is the same as before, I just touched the dump function.

laszlodaniel, 2020-07-23:
It's cool, his fuel gauge IC is acting strange and now he's investigating.

youxiaojie, 2020-07-23:
I have tested again , it is ok, 17 can be changed.

```
[INFO] Connecting to COM22

[<-TX] Handshake request (COM22)
3D 00 02 01 00 03

[RX->] Handshake response
3D 00 08 81 00 53 42 48 41 43 4B 35

[INFO] Handshake OK: SBHACK

[INFO] Device connected (COM22)

[RX->] Device settings
3D 00 0B 83 01 03 2B 5C 12 C0 08 24 01 33 4B

[INFO] Word byte-order: reverse read/write

[INFO] Design voltage: 11.1 V Design capacity: 4800 mAH

[INFO] Chip: BQ2084 Firmware ver 1.33

[<-TX] Scan SMBus
3D 00 02 02 02 06

[RX->] Scan SMBus address result
3D 00 03 82 02 0B 92

[INFO] SMBus device(s): 0B

[<-TX] SMBus register dump request
3D 00 04 02 03 00 FF 08

[RX->] Error: not enough MCU RAM
3D 00 03 8F FD FF 8E

[<-TX] SMBus register dump request
3D 00 04 02 03 00 3F 48

[RX->] SMBus register dump (00-3F)
3D 00 DA 82 03 00 3F 00 01 33 01 01 E0 02 00 0A 03 60 81 04 00 00 05 FF FF 06 FF FF 07 00 01 08 
0B D0 09 2D AD 0A 00 00 0B 00 00 0C 00 08 0D 00 64 0E 00 64 0F 12 C0 10 12 C0 11 FF FF 12 FF FF 
13 FF FF 14 03 20 15 31 38 16 40 E0 17 00 00 18 12 C0 19 2B 5C 1A 00 31 1B 35 50 1C 06 9B 1D 06 
9B 1E 06 9B 1F 06 9B 20 03 45 43 53 21 06 33 32 30 43 31 32 22 04 4C 49 4F 4E 23 0D 06 58 1B B8 
67 01 00 11 00 C8 00 63 2A 24 00 00 25 00 00 26 00 00 27 00 00 28 00 00 29 00 00 2A 00 00 2B 00 
00 2C 00 00 2D 00 00 2E 00 00 2F 06 90 30 06 90 31 06 90 32 06 90 33 06 90 34 06 90 35 06 90 36 
06 90 37 06 90 38 06 90 39 06 90 3A 06 90 3B 06 90 3C 00 00 3D 0F 0F 3E 0F 4A 3F 0F 55 B7

[INFO] SMBus register dump details (00-3F):
[00]: 01 33 // ManufacturerAccess: 01 33
[01]: 01 E0 // RemainingCapacityAlarm: 480 mAh
[02]: 00 0A // RemainingTimeAlarm: 10 minuets
[03]: 60 81 // BatteryMode: Report in mA or mAh (default). Disable ChargingVoltage and ChargingCurrent broadcasts to Charger. Disable AlarmWarning broadcasts to Host and Charger. Capacity Re-Learn Cycle Requested. Primary or Secondary Battery Not Supported. Internal Charge Control Disabled (default).
[04]: 00 00 // AtRate: 0 mAh
[05]: FF FF // AtRateTimeToFull: 65535 minuets
[06]: FF FF // AtRateTimeToEmpty: 65535 minuets
[07]: 00 01 // AtRateOK: True
[08]: 0B D0 // Temperature: 29.4 °C
[09]: 2D AD // Voltage: 11693 mV
[0A]: 00 00 // Current: 0 mA
[0B]: 00 00 // AverageCurrent: 0 mA
[0C]: 00 08 // MaxError: 8%
[0D]: 00 64 // RelativeStateOfCharge: 100%
[0E]: 00 64 // AbsoluteStateOfCharge: 100%
[0F]: 12 C0 // RemainingCapacity: 4800 mAh
[10]: 12 C0 // FullChargeCapacity: 4800 mAh
[11]: FF FF // RunTimeToEmpty: 65535 minuets
[12]: FF FF // AverageTimeToEmpty: 65535 minuets
[13]: FF FF // AverageTimeToFull: 65535 minuets
[14]: 03 20 // ChargingCurrent: 800 mA
[15]: 31 38 // ChargingVoltage: 12600 mV
[16]: 40 E0 // BatteryStatus: ALARMS: TERMINATE_CHARGE STATUS: INITIALIZED DISCHARGING FULLY_CHARGED 
[17]: 00 00 // CycleCount: 0 Cycles
[18]: 12 C0 // DesignCapacity: 4800 mAh
[19]: 2B 5C // DesignVoltage: 11100 mV
[1A]: 00 31 // SpecificationInfo: Smart Battery Spec 1.1 with PEC
[1B]: 35 50 // ManufactureDate: Y2006M10D16
[1C]: 06 9B // SerialNumber: 1691
[1D]: 06 9B // Reserved: 1691
[1E]: 06 9B // Reserved: 1691
[1F]: 06 9B // Reserved: 1691
[20]: 45 43 53 // ManufacturerName: ECS
[21]: 33 32 30 43 31 32 // DeviceName: 320C12
[22]: 4C 49 4F 4E // DeviceChemisty: LION
[23]: 06 58 1B B8 67 01 00 11 00 C8 00 63 2A // ManufacturerData: 06 58 1B B8 67 01 00 11 00 C8 00 63 2A
[24]: 00 00 // Reserved: 0
[25]: 00 00 // Reserved: 0
[26]: 00 00 // Reserved: 0
[27]: 00 00 // Reserved: 0
[28]: 00 00 // Reserved: 0
[29]: 00 00 // Reserved: 0
[2A]: 00 00 // Reserved: 0
[2B]: 00 00 // Reserved: 0
[2C]: 00 00 // Reserved: 0
[2D]: 00 00 // Reserved: 0
[2E]: 00 00 // Reserved: 0
[2F]: 06 90 // OptionalMfgFunction5: HEX_0690 BIN_00000110_10010000 DEC_1680 Or a 144-byte block. BQ208X: Pack Status and Pack Configuration: System present. V>EndDischargeVoltage2. Unsealed. Discharge cycle valid for an FCC update.AFE com OK. PF Flag clear. CellVoltage OK. 
[30]: 06 90 // Reserved: 1680
[31]: 06 90 // Reserved: 1680
[32]: 06 90 // Reserved: 1680
[33]: 06 90 // Reserved: 1680
[34]: 06 90 // Reserved: 1680
[35]: 06 90 // Reserved: 1680
[36]: 06 90 // Reserved: 1680
[37]: 06 90 // Reserved: 1680
[38]: 06 90 // Reserved: 1680
[39]: 06 90 // Reserved: 1680
[3A]: 06 90 // Reserved: 1680
[3B]: 06 90 // Reserved: 1680
[3C]: 00 00 // OptionalMfgFunction4 (VCell4 for BQ): 0 mV
[3D]: 0F 0F // OptionalMfgFunction3 (VCell3 for BQ): 3855 mV
[3E]: 0F 4A // OptionalMfgFunction2 (VCell2 for BQ): 3914 mV
[3F]: 0F 55 // OptionalMfgFunction1 (VCell1 for BQ): 3925 mV

[<-TX] Write word data
3D 00 05 05 02 17 00 01 24

[RX->] Word data write response
3D 00 06 85 02 17 00 01 02 A7

[INFO] Reg.: 17
       Data: 00 01
       # of bytes written: 02

[<-TX] Read word data
3D 00 03 04 02 17 20

[RX->] Word data received
3D 00 05 84 02 17 00 01 A3

[INFO] Reg.: 17
       Data: 00 01
```

youxiaojie, 2020-07-23:
again, 17 changed to 0a,and back to 00.
```
[<-TX] Write word data
3D 00 05 05 02 17 00 00 23

[RX->] Word data write response
3D 00 06 85 02 17 00 00 02 A6

[INFO] Reg.: 17
       Data: 00 00
       # of bytes written: 02

[<-TX] SMBus register dump request
3D 00 04 02 03 17 19 39

[RX->] SMBus register dump (17-19)
3D 00 0D 82 03 17 19 17 00 00 18 12 C0 19 2B 5C 63

[INFO] SMBus register dump details (17-19):
[17]: 00 00 // CycleCount: 0 Cycles
[18]: 12 C0 // DesignCapacity: 4800 mAh
[19]: 2B 5C // DesignVoltage: 11100 mV
```

youxiaojie, 2020-07-23:
try to read 17 by block, to see is it a 22 bytes block. because if you read a block by work, the most byte will be length of the block. so 0x16 maybe the lenth of the block.

youxiaojie, 2020-07-23:
could you dump register 00-3f? and 40 to ff?

youxiaojie, 2020-07-23:
datasheet of a2168
i2c address 0x16
https://wenku.baidu.com/view/55007043b90d6c85ec3ac681.html

youxiaojie, 2020-07-23:
If you are willing, git reset to origianal version, to see is it ok?

menjic, 2020-07-24:
> datasheet of a2168
> i2c address 0x16
> what if you write 0001 to reg 0 and read from reg0?
> what if you write 0002 to reg 0 and read from reg0?
> https://wenku.baidu.com/view/55007043b90d6c85ec3ac681.html
> 
> https://appwk.baidu.com/naapi/doc/view?ih=918&o=jpg_6&iw=1188&ix=0&iy=0&aimw=1188&rn=1&doc_id=e228f2075ef7ba0d4b733ba6&pn=1&sign=5e537eecebc9be0930bb27ddc6546776&type=1&app_ver=2.9.8.2&ua=bd_800_800_IncredibleS_2.9.8.2_2.3.7&bid=1&app_ua=IncredibleS&uid=&cuid=&fr=3&Bdi_bear=WIFI&from=3_10000&bduss=&pid=1&screen=800_800&sys_ver=2.3.7

I am not brave enought to write anywhere where I am not sure I would not cause something I do not understand, so for now I tested write to cycle count and batt capacity (0x17 and 0x18) in both cases I can write byte but can not write word, what ever I try to write with word write, when I read back I get 00 16 which is strange I did not have much time to play with this but I will, I see you can write to BQxxx this mine chip a2168 is very similar but obviously not same identical as BQxxx.
So sum:
Read:
read byte 0x17 = 00
read word 0x17 = 00 16

Write case 1:
- write byte 0A to 0x17
- read byte 0x17 = 16
- read word 0x17 = 0A 16 

Write case 2:
- write word 00 00 to 0x17
- read byte 0x17 = 16
- read word 0x17 = 0A 16

Write case 3:
- write byte 00 to 0x17
- read byte 0x17 = 16
- read word 0x17 = 00 16

As you can see if I explained right, writing byte works but it writes first byte in the word and as I can see 0x17 should be word not block, it is cycle count and before I tested writing cycle count was 00 00, when I dump register 0x17 I saw that and battery is new never used I bought it to play with so this was expected.

First time I tried to write I got 00 16 and now can not do anything with that second byte.

First byte I can write but only with write byte command not with write word command.

when I read byte no mater what is chosen reverse or not read byte command gives me result 16.

when I write word nothing changes.

when I write byte and read word I can see first byte is written to what ever I want.

menjic, 2020-07-24:
I have bunch of other batteries and will test with them I just did not have time to do that, I suspect that this chip requires some different settings compared to BQxx but I do not understand all that well to know what is not correct or is that the problem or somethin totally different. I don't believe chip is broken but that also could be the case and lack of my experience is preventing me to see that, but battery is charging and discharging and reporting results correctly so I don't know where and what I am doing wrong.

menjic, 2020-07-24:
**Quick update**

Software  BE2Works can write words, bytes, blocks, to smbus with ebay CP2112 usb2smbus hid board.

Tested with data I was unable to write to my battery with A2168 chip inside, with Arduino and Smart Battery Hack GUI,

Connected my battery to CP2112, started BE2Works, clicked to send smbus commands, manual is present on their website, written words for capacity and cycle count, disconnected, connected to Smart Battery Hack and Arduino, reading values I just wrote all seem ok. Tested block writing in same whey, tested byte writing in same whey all works as expected.

So something is different for this chip A2168 compared BQxx chips in terms of writing data or again I am doing something very wrong.

If anyone has some Ideas please feel free to share.

BE2Works is not completely free app but it is useful but I would like to fix this one because it is much better and modern and has more potential and I will continue to think about this and test and inform you if I find something.

youxiaojie, 2020-07-24:
> **Quick update**
>
> Software BE2Works can write words, bytes, blocks, to smbus with ebay CP2112 usb2smbus hid board.
>
> Tested with data I was unable to write to my battery with A2168 chip inside, with [...]

you can read and write with be2works correctly? but smart battery hack could not read write correctly?

do you see the datasheet p33
 I see it mentioned modify parameter.0x77 to indicate sector, and 32bytes read/write each time. eg, in p54 cycle count, sector3 offset 18. so write 0003 to 0x77 to set sector, to usethe extended sbs command 0x78 in p82 to read 32byte a time, 
find value in offset 18.

youxiaojie, 2020-07-24:
do you have the unseal key for a2168? when using be2works?

menjic, 2020-07-24:
> do you have the unseal key for a2168? when using be2works?

No I dont need it because my battery was factory unsealed.

menjic, 2020-07-24:
Smart Battery Hack can read everything correctly.

Can not write word.

And write byte I am not sure is correct but it does write byte.

Write block I did not try because I am afraid to not mess something big until I solve byte and word write.

menjic, 2020-07-24:
I read dayasheet p33 and all other pages few times, I am not sure I do understand it corectly. So I will read it few more times.

I saw that for reading whole block but then how BE2Works can wride word and block if it is needed to do procedure you described ?

BE2Works does not know what chip this is and what are procedures for writing I thin, I may be wrong.

As I see BE2Works is sending simple smbus commands with data you provide but obviously does something different than Smart Battery Hack, and I will find what dont worry :) If you dont before me :)

laszlodaniel, 2020-07-25:
Okay, so the datasheet makes it clear that simple byte and word write won't work. You have to use 8 block reads and 8 block writes instead (256 bytes in total) to update whole flash sectors, even if you want to change 1 byte only.

First you have to find the sector in which your data resides and select it for reading. In the datasheet the Term Voltage variable is in Sector 2:

```
Select register: 0x77 in the write line.
Write: 0x0002, click word button.
```

Then you have to block read registers between 0x78 and 0x7F and save their values in a text editor in separate lines:

```
Select register: 0x78 in the read line.
Click block button to read 32 bytes.
Copy 32 bytes result to text editor.

Select register: 0x79 in the read line.
Click block button to read 32 bytes.
Copy 32 bytes result to text editor.

...

Select register: 0x7F in the read line.
Click block button to read 32 bytes.
Copy 32 bytes result to text editor.
```

The first byte value in these registers should be the block length (0x20) so the block read command will know to read exactly 32 bytes.
Now you have 8 lines of 32 bytes in the text editor. This is a whole flash sector containing various variables including the ones you want to change. Make your changes carefully.

Then you have to select the sector again for writing with the same command as in the beginning:

```
Select register: 0x77 in the write line.
Write: 0x0002, click word button.
```

Then you have to block write registers between 0x78 and 0x7F with the modified lines in your text editor:

```
Select register: 0x78 in the write line.
Copy modified 32 bytes line to the write textbox.
Click block button to write 32 bytes.

Select register: 0x79 in the write line.
Copy modified 32 bytes line to the write textbox.
Click block button to write 32 bytes.

...

Select register: 0x7F in the write line.
Copy modified 32 bytes line to the write textbox.
Click block button to write 32 bytes.
```

Verify update by byte/word reading in the appropriate registers.

menjic, 2020-07-26:
laszlodaniel all that is pretty complicated, and I am not saying that it is wrong, I saw it is written in datasheet, but if you take into account that I wrote just word, just byte and just word with BE2Works that does not seem to reflect datasheet and it is strange unless that software has integrated this chip specifically. I literally used their command "ww 17 0000" for example that is write word at 0x17 data is 0000

As I understand this all, if simple command like this can write word in that software and datasheet procedure is to be respected, it would have to know which chip is this and do that 8 block read, replace my data in given sector and write back all.

I don't have oscilloscope or another why to physically check what is happening when I try to write but I am working on finding someone close to me, then I hope I will be able to see how exactly BE2Works writes and what.

I am sorry to take your time on this, I am not any expert in microcontrollers and I may be wrong in 100 things here so dont pay much attention on this for now :)

And thank you for title, I should have to do it in the beginning I am sorry.

laszlodaniel, 2020-07-27:
No worries. The datasheet suggests that reading/writing configuration data can only be done in blocks (32 bytes). You can get away with 1 block reads once you set the sector address but you need to write the whole sector (8x32 bytes) if you want to modify something.

BE2Works can easily do this internally without the user noticing. So there's no surprise here. This is just how flash writing works for this chip.

This method doesn't considers regular SBS commands where the chip returns single bytes and words. Again reading flash is always easier.

I think the fact that you can write individual bytes and words and sometimes succeed is just some kind of firmware-related bug.

Before you dive in and write whole sectors it would be a good idea to check if there's a checksum byte/word somewhere in the sectors. BE2Works would do checksum calculation if this chip requires it, although I don't see checksum mentioned in the datasheet.

So first do a whole sector read with SmartBatteryHack_GUI and save it as a backup.
Then change a single variable with BE2Works and read the sector again with SmartBatteryHack_GUI and save it.
Finally compare the two sector and see if any other data is changed apart from the variable.

This whole process can be automated in the SmartBatteryHack Arduino code once it's certain that this is the only way to go.

menjic, 2020-07-27:
I will do that, good idea, thank you. Bugs are good for us hahaha I hope it is and there is more :)
In few places I found mentions about writing problems in other chips with other software and people solving it by repeatedly writing and reading for comparison until chip accepts data you want to write, if I remember correctly they say it is needed to write 3 to 5 times possibly bug also or bad interface or design of something but they say it worked like that. It is interesting option for test also.
Problem is that this is literally endless area and our time is limited but I do enjoy in problems and unknown stuff :)

lmdpua, 2020-12-16:
> Select register: 0x77 in the write line.
> Write: 0003, click word button.

Hello! It's clear. But I have the same problem.
If I read address 0x77 in a new battery there is 0x00 00
If I write to 0x77 0x00 03, I read 0x00 00, but if I write 0x03 byte, I read 0x03 16
Write 0x01 Read 0x01 16
And then whatever value was written in 0x77, in 0x78, the same data all the time
very strange chip

lmdpua, 2021-01-14:
I bought a logic analyzer to deal with this problem. 
[image]
[image]
As you can see in the screenshots, be2works writes words, while the arduino library writes two bytes. This is suitable for some batteries. But for others it is necessary write word.
What do you think about this?

lmdpua, 2021-01-14:
And this screenshot with #define I2C_TIMEOUT 1
[image]

laszlodaniel, 2021-01-15:
I don't see anything unusual, except that the BE2Works SMBus clock signal has more than 50% duty cycle.
The fundamental data units are bytes and when words are written there's a small delay between the bytes to be expected.
Are you referring to the "Decoded Protocols" showing bytes instead of words?

lmdpua, 2021-01-17:
I seem to have solved the problem! For the fiftieth time looking at these screenshots, I thought, but not to look into the [smbus specification.](http://smbus.org/specs/smbus110.pdf)
[image]
That is, the order must be followed. And in our case, it is not observed.
[image]
We have an additional function call `i2c_rep_start`. That spoils everything. I commented out this line and lo and behold! Everything worked out! The word is being written!
iiiihhhhaaa!

laszlodaniel, 2021-01-17:
Try the new Arduino code as well. I switched to the standard Wire library that handles data reading/writing properly.

lmdpua, 2021-01-17:
Yes, I will definitely try. But right now I have a battery tester in development. I'll show you the unfinished part of the project, how it looks. The body and everything inside is almost finished, and I'm too on the verge of redoing the project now

lmdpua, 2021-01-17:
I made a project based on a Chinese NMI display. The interface is still only in Russian, but I will add English later, this is not the biggest problem. In the video, I change the number of cycles. The battery is Chinese, in the unseal mode
[My laptop battery tester](https://www.youtube.com/watch?v=3IRjtnhbxkE)

youxiaojie, 2021-01-20:
> I seem to have solved the problem! For the fiftieth time looking at these screenshots, I thought, but not to look into the [smbus specification.](http://smbus.org/specs/smbus110.pdf)
> [image]
> That [...]

Well done!

jaro44, 2021-03-10:
> 
> 
> Try the new Arduino code as well. I switched to the standard Wire library that handles data reading/writing properly.

it doesn't work properly with A2168, old arduino sketch with fix from lmdpua work ok,
I have successfully changed: 
- CycleCount = 0 (now is 1 after 1 full cycle) 
- DesignCapacity = 4800 mAh 
- and RemainingCapacityAlarm = 480 mAh ,
 but i have a problem with FullChargeCapacity:
```
[RX->] SMBus register dump (00-18)
3D 00 4F 82 03 00 18 00 90 06 01 01 E0 02 00 0A 03 60 01 04 00 00 05 FF FF 06 FF FF 07 00 01 08 
0B 6F 09 2C E1 0A 00 00 0B 00 10 0C 00 02 0D 00 23 0E 00 04 0F 00 D3 10 02 5A 11 FF FF 12 FF FF 
13 FF FF 14 0B B8 15 31 38 16 02 C0 17 00 01 18 12 C0 A1

[INFO] SMBus register dump details (00-18):
[00]: 90 06 // ManufacturerAccess: 90 06
[01]: 01 E0 // RemainingCapacityAlarm: 480 mAh = 31457 mWh
[02]: 00 0A // RemainingTimeAlarm: 10 minutes
[03]: 60 01 // BatteryMode: 0110000000000001
[04]: 00 00 // AtRate: 0 minutes
[05]: FF FF // AtRateTimeToFull: 65535 minutes
[06]: FF FF // AtRateTimeToEmpty: 65535 minutes
[07]: 00 01 // AtRateOK: true
[08]: 0B 6F // Temperature: 26,54°C
[09]: 2C E1 // Voltage: 11,489 V
[0A]: 00 00 // Current: 0 A
[0B]: 00 10 // AverageCurrent: 0,016 A
[0C]: 00 02 // MaxError: 2%
[0D]: 00 23 // RelativeStateOfCharge: 35%
[0E]: 00 04 // AbsoluteStateOfCharge: 4%
[0F]: 00 D3 // RemainingCapacity: 211 mAh
[10]: 02 5A // FullChargeCapacity: 602 mAh
[11]: FF FF // RunTimeToEmpty: 65535 minutes
[12]: FF FF // AverageTimeToEmpty: 65535 minutes
[13]: FF FF // AverageTimeToFull: 65535 minutes
[14]: 0B B8 // ChargingCurrent: 3 A
[15]: 31 38 // ChargingVoltage: 12,6 V
[16]: 02 C0 // BatteryStatus: 0000001011000000
[17]: 00 01 // CycleCount: 1
[18]: 12 C0 // DesignCapacity: 4800 mAh
```
I replaced my cells with a 3x 2400 mAh samsung 
battery is charged to 100%
then it quickly drops to 0% and works after about 2 hours 
am i doing something wrong?

laszlodaniel, 2021-03-10:
Do you write the FCC word the same way like the rest of the settings?

laszlodaniel, 2021-03-10:
What if you write a smaller value? Perhaps maximum value is limited by firmware.

jaro44, 2021-03-10:
is same
```
[INFO] Reg.: 10
       Data: 01 E0
       # of bytes written: 02

[<-TX] Read word data
3D 00 03 04 02 10 19

[RX->] Word data received
3D 00 05 84 02 10 02 5A F7

[INFO] Reg.: 10
       Data: 02 5A
```

laszlodaniel, 2021-03-10:
According to the A2168 datasheet the settings you were able to change are R/W when the chip is in unsealed/full access mode. However Full Charge Capacity register remains readable only.

To change FCC you have to manipulate whole flash-blocks. Do you have the datasheet? There's a step-by-step instruction on how to do this.

jaro44, 2021-03-10:
no, I don't :(

menjic, 2021-03-10:
I have it I will send it here.

laszlodaniel, 2021-03-10:
Here you go:
https://mega.nz/file/PFkGGapR#2bpnEshy_TqWoIRRCXpOGl3yvDAtjCzRhdmEIDQBcgk

Starting from page 63 you can see regular SBS commands. Some have a note that those words are R/W when the chip is unsealed or in full access mode, like cycle count, design capacity and so forth. Unfortunately FCC doesn't have this note, so it's not that easy to change.

On page 32 there's the Configuration data chapter, where reading and writing variables are explained. An example is given too. Basically you have to block read and write 32 bytes of data. Because the flash structure you can't change 2 bytes only, you have write full 32 byte blocks (containing your change).
I don't know if blocks are protected by a checksum, so first read the necessary blocks and backup them.

laszlodaniel, 2021-03-10:
Alright.

jaro44, 2021-03-10:
If anyone could show me how to go about it, I would risk trying.
In this state the battery cannot be used anyway (frequent system messages about the battery).
Thanks

jaro44, 2021-03-10:
how I can read the necessary sectors and backup them?

menjic, 2021-03-10:
0x10 says it is read-word, not read-write, can you try to empty cells or
battery in use and then charge it to full installed in the device and see
would that work there is possibility that chip will correct that value
after full charge ??

I am just guessing and thinking I don't know can that work but it does not
hurt much to test :)

menjic, 2021-03-10:
> no it doesn't help, FCC is always 602mAh

I will look into my chip to see what happens later today or tonight and
inform you what I find out.

jaro44, 2021-03-10:
Thank You, i am also trying to find something on the net about it.

jaro44, 2021-03-11:
After an all-night fight, I could not do anything with this battery with the a2168 chip :(
But I disarmed the old battery (is not recognized by the system) and there is a bq8050 chip
If I can do something with it, it seems to be locked?
```
[INFO] Connecting to COM3

[<-TX] Handshake request (COM3)
3D 00 02 01 00 03

[RX->] Handshake response
3D 00 08 81 00 53 42 48 41 43 4B 35

[INFO] Handshake OK: SBHACK

[INFO] Device connected (COM3)

[RX->] Device settings
3D 00 0B 83 01 03 2A 30 11 94 03 2C 03 2C EF

[INFO] Word byte-order: reverse read/write

[INFO] Design voltage: 10,8 V

[<-TX] Scan SMBus
3D 00 02 02 02 06

[RX->] Scan SMBus address result
3D 00 03 82 02 0B 92

[INFO] SMBus device(s): 0B

[<-TX] Select SMBus address
3D 00 03 03 02 0B 13

[RX->] SMBus settings
3D 00 03 83 02 0B 93

[INFO] Current SMBus device address: 0B

[<-TX] SMBus register dump request
3D 00 04 02 03 00 21 2A

[RX->] SMBus register dump (00-21)
3D 00 79 82 03 00 21 00 03 2C 01 01 C2 02 00 0A 03 00 00 04 00 00 05 FF FF 06 FF FF 07 FF FF 08 
08 A7 09 2B 10 0A 00 00 0B 00 00 0C 00 14 0D 00 00 0E 00 00 0F 00 00 10 06 4D 11 00 00 12 00 00 
13 FF FF 14 00 00 15 00 00 16 53 D0 17 00 C3 18 11 94 19 17 17 1A 00 21 1B 3E E7 1C 2B 6D 1D 17 
17 1E 17 17 1F 17 17 20 05 53 41 4E 59 4F 21 0C 4E 53 32 50 33 53 5A 4D 50 34 57 52 CE

[INFO] SMBus register dump details (00-21):
[00]: 03 2C // ManufacturerAccess: 03 2C
[01]: 01 C2 // RemainingCapacityAlarm: 450 mAh = 4860 mWh
[02]: 00 0A // RemainingTimeAlarm: 10 minutes
[03]: 00 00 // BatteryMode: 0000000000000000
[04]: 00 00 // AtRate: 0 minutes
[05]: FF FF // AtRateTimeToFull: 65535 minutes
[06]: FF FF // AtRateTimeToEmpty: 65535 minutes
[07]: FF FF // AtRateOK: true
[08]: 08 A7 // Temperature: 19,42°C
[09]: 2B 10 // Voltage: 11,024 V
[0A]: 00 00 // Current: 0 A
[0B]: 00 00 // AverageCurrent: 0 A
[0C]: 00 14 // MaxError: 20%
[0D]: 00 00 // RelativeStateOfCharge: 0%
[0E]: 00 00 // AbsoluteStateOfCharge: 0%
[0F]: 00 00 // RemainingCapacity: 0 mAh
[10]: 06 4D // FullChargeCapacity: 1613 mAh
[11]: 00 00 // RunTimeToEmpty: 0 minutes
[12]: 00 00 // AverageTimeToEmpty: 0 minutes
[13]: FF FF // AverageTimeToFull: 65535 minutes
[14]: 00 00 // ChargingCurrent: 0 A
[15]: 00 00 // ChargingVoltage: 0 V
[16]: 53 D0 // BatteryStatus: 0101001111010000
[17]: 00 C3 // CycleCount: 195
[18]: 11 94 // DesignCapacity: 4500 mAh
[19]: 17 17 // DesignVoltage: 5,911 V
[1A]: 00 21 // SpecificationInfo: 0000000000100001
[1B]: 3E E7 // ManufactureDate: 2011.07.07
[1C]: 2B 6D // SerialNumber: 2B 6D
[1D]: 17 17 // 17 17
[1E]: 17 17 // 17 17
[1F]: 17 17 // 17 17
[20]: 05 53 // ManufacturerName: 05 53
[21]: 4E 59 // DeviceName: 4E 59
[22]: 21 0C // DeviceChemistry: 21 0C
[23]: 53 32 // ManufacturerData: 53 32
[24]: 33 53 // 33 53
[25]: 4D 50 // 4D 50
[26]: 57 52 // 57 52

[INFO] Device disconnected (COM3)
```

laszlodaniel, 2021-03-11:
> 
> 
> how I can read the necessary sectors and backup them?

Write Sector ID (0):
```
77 0000 write word
```

Read Sector blocks (32 bytes):
```
78 read block -> copy 32 bytes result to notepad
79 read block -> copy 32 bytes result in a new line
7A read block
7B read block
7C read block
7D read block
7E read block
7F read block
```

Now move on to Sector ID 1:
```
77 0001 write word
```

Read Sector blocks again (32 bytes):
```
78 read block -> copy 32 bytes result to notepad
79 read block -> copy 32 bytes result in a new line
7A read block
7B read block
7C read block
7D read block
7E read block
7F read block
```

Continue as long as valid sector pages are returned. There's at least 3 sectors (0, 1, 2).

Your backup should look something like this:

```
Sector ID 0:
01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF
01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF
01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF
01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF
01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF
01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF
01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF
01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF

Sector ID 1:
01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF
01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF
01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF
01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF
01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF
01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF
01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF
01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF

Sector ID 2:
01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF
01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF
01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF
01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF
01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF
01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF
01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF
01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF 01 23 45 67 89 AB CD EF
```

I can't find the exact offset where FCC is stored so you have to look through all this data and look for this word:

```
02 5A
```

>
>
> bq8050 chip. If I can do something with it, it seems to be locked?

There's no public information available on how to unlock this chip. The basic idea behind unsealing is similar to BQ8030:

http://www.karosium.com/2016/08/hacking-bq8030-with-sanyo-firmware.html

but the code challenge is rolling continuously and won't let us through.

jaro44, 2021-03-11:
ok, Thank You very much,
in the evening and maybe at night I'll fight it again.

jaro44, 2021-03-11:
doesn't find "02 5A", unless it is to be reversed 

```
0000:
3D 00 24 84 03 78 20 10 04 0F A0 10 CC 10 04 10 CC 0F 3C 02 0A {5A 02} 0B B8 13 88 02 00 C8 23 28 02 00 C8 13 88 02 23 82
3D 00 24 84 03 79 20 28 02 08 0A 0F 00 05 77 77 02 EE 02 02 8A 02 EE 02 02 8A 02 EE 02 02 8A 02 F3 02 02 8A 02 F3 11 85

0001:
3D 00 24 84 03 78 20 02 31 00 0A 00 81 2A 30 00 21 4E BE CF 3C 00 00 10 40 5A 64 14 50 15 F0 05 53 41 4E 59 4F 00 00 99
3D 00 24 84 03 79 20 00 00 00 00 00 00 00 00 00 00 00 00 00 0C 4E 53 32 50 33 53 5A 4D 50 34 57 52 00 00 00 00 00 00 CD

0002:
3D 00 24 84 03 78 20 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 4A
3D 00 24 84 03 79 20 BC 95 3B 72 E6 D3 04 EC 00 00 0F 96 06 A9 BD D4 30 29 ED DB 00 00 17 69 EF 14 22 03 00 23 28 00 E4

0003:
3D 00 24 84 03 78 20 27 10 CB A8 27 10 C3 43 CB A8 C3 43 00 00 00 03 BC EB 00 02 00 26 00 00 00 02 01 2C FE 63 27 10 3C
3D 00 24 84 03 79 20 16 1F 27 10 0A 4F 07 39 FF B3 64 03 FF 00 00 0C 9E 0C 1C FF FF FF FF FF FF FF FF FF FF FF FF FF 26

0004:
3D 00 24 84 03 78 20 FF FF FF FF FF FF 00 00 0F 2E 00 01 00 00 0F 30 FF FA 00 00 0F 46 FF F9 00 00 0F 23 FF FD 00 00 2E
3D 00 24 84 03 79 20 97 50 00 E7 00 00 97 40 01 24 FF FF FF FF FF FF 00 00 04 8C 00 02 FF FF FF FF FF FF FF FF FF FF 90

0005:
3D 00 24 84 03 78 20 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 23
3D 00 24 84 03 79 20 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 24

0006:
3D 00 24 84 03 78 20 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 23
3D 00 24 84 03 79 20 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 24

```

laszlodaniel, 2021-03-11:
It's probably due to little endianness, I think that's it.
What about the rest of pages of the sectors? Are they just FFs?
Don't copy the full usb packet, just the 32 bytes payloads.

[73 more replies]