laszlodaniel/SmartBatteryHack #11: Documentation for BQ8050

Nicarim, 2020-07-22:
I've read your blog on your attempt on BQ8050
https://boundarycondition.home.blog/2020/01/18/the-repairing-and-hacking-of-a-dell-j1knd-bq8050-laptop-battery/

And I dig through baidu and found this - do you think it might be any helpful?

http://www.docin.com/p-859124464.html

Nicarim, 2020-07-22:
@youxiaojie hello I've seen you wer active around here and you're from china - would you be so kind to download it from docin.com I linked above and share it with me/us/here so it can be downloaded without registering? I cannot register there because I don't understand a thing 😞

laszlodaniel, 2020-07-22:
Excellent find, thank you!

These documents seem to be behind a paywall but I have found a bypass here:
https://dearplain.github.io/

Enter link at the top and press the button:
http://www.docin.com/p-859124464.html

Pages are loading slowly and new page is rendered when you scroll all the way down. There's no PDF download but I'll make one from the images and upload it soon.

Nicarim, 2020-07-22:
Glad I could help - I've got a battery with this exact chip currently and was digging into it using what you've explained in your blog. What I didn't get yet from there is - were you able to extract eeprom?

laszlodaniel, 2020-07-23:
Datasheet uploaded!
I was able to read some kind of memory but I'm not sure what it is. It seems to be containing repeating values. The GUI has a memory reader in the Debug group. I labeled there 3 distinct ranges where unique data resides.
Unfortunately I couldn't read the same values twice. Someone else uploaded their eeprom dump to Google Drive and linked it in the blog post's comments.
We need to inspect the datasheet and see if there's anything that may help.

Nicarim, 2020-07-23:
From a quick glance there is information where each memory regions are, how SHA1 authorization works and default key is also provided (although unlikely to be that?). What boggles my mind is how SMBus maps to these memory regions and if it does at all since this is handled by the firmware (or is it?). It is not that simple to just ask for given region through smbus and it just returns it - is it? 

I'm just beginning the hacking journey so I might miss some fundamental pieces

laszlodaniel, 2020-07-25:
Well, reading memory blocks seem to be easy even in sealed mode:

```
#define SetROMAddress               0x40 // word write only
#define PeekROMByte                 0x42
#define PeekROMBlock                0x43 // block read, size seems to be always 0x20 (32 bytes)
```

With these registers you can read ROM bytes. The Read ROM button in the GUI uses these registers and 3 distinct intervals to read from.
I think the SHA1 authorization is for write related commands. And according to the datasheet the firmware that does the calculations can only be written once fresh then it can't be changed again. That leaves us with nothing dangerous in our hands.

When I ordered my replacement battery (not BQ8050) it came in unsealed (possibly full access) mode and I saw and saved all the security keys, which were all default ones. So I wouldn't be surprised if pack makers would leave default keys untouched. Only way to find out is to go through the authentication procedure, give the chip a random challenge, assume default encryption keys and see if in theory the chips response can be decrypted with said default keys.

Nicarim, 2020-07-25:
Thank you very much for detailed information - if I work anything out I'll be adding more information here :)