laszlodaniel/SmartBatteryHack #13: Need help with BQ30z55

bacrua, 2020-10-04:
Dear friend. Thank you for your hard work. I read the protocol for b30z55, a lot is not clear, I tried to do it as written in your documentation, it did not work out. Help me figure it out and make it unseal.

laszlodaniel, 2020-10-06:
This chip requires SHA-1 authentication which is out of the scope of this project. It needs sophisticated algorithms and automated functions. This project handles simple manual word writes only that are enough for unsealing certain old chips.

https://www.ti.com/lit/ug/sluua79/sluua79.pdf
On page 53 there's a step-by-step description to authenticate/unseal this chip. It's rather tedious. I would imagine there's a timeout between the steps but if not you can try doing manual calculations.

bacrua, 2020-10-06:
@laszlodaniel
 thanks, I read this. For SHA-1 you need an input password. On the forums they write that many batteries have a standard password. But which one? Nobody writes and does not speak.
Thank you, your project is pretty cool. I only have one battery for the old HP. Then I have no desire to buy a paid program, but I am interested in finding out and fixing it. I have some knowledge in programming and if the "fire of desire" does not go out, I will try to unblock it with the help of the program.

laszlodaniel, 2020-10-06:
You're right about default keys, manufacturers usually don't bother changing it.
The default authentication key is 0x0123456789ABCDEFFEDCBA9876543210.

To get going try the first two steps where no calculation is needed.
The first word writes may need to be reverse ordered if there's nothing to read at the second step. This is trial and error.
At the second step you need to use block reading to get messages longer than 2 bytes.

```
1. Send Unseal (0x0031) or Full Access (0x0032) command to ManufacturerAccess (0x00).
2. Read 160-bit message M from ManufacturerInput (0x2F) in the format 
   0xAABBCCDDEEFFGGHHIIJJKKLLMMNNOOPPQQRRSSTT, where AA is LSB.
3. Generate SHA-1 input block B1 of 512 bytes (total input =128-bit unseal/full access key KD + 
   160 bit message M + 1 + 159 0s + 100100000).
4. Generate SHA-1 hash HMAC1 using B1.
5. Generate SHA-1 input block B2 of 512 bytes (total input =128-bit unseal/full access key KD + 
   160 bit hash HMAC1 + 1 + 159 0s + 100100000).
6. Generate SHA-1 hash HMAC2 using B2.
7. Write 160-bit hash HMAC2 to ManufacturerInput() in the format 
   0xAABBCCDDEEFFGGHHIIJJKKLLMMNNOOPPQQRRSSTT, where AA is LSB.
8. Device compares hash HMAC2 with internal calculated hash HMAC3. If it matches, device allows 
   UNSEALED/FULL ACCESS mode indicated with the OperationStatus()[SEC1],[SEC0] flags.
```

bacrua, 2020-10-06:
OK. Thanks again. I will stay. I will report the result.

bacrua, 2020-10-06:
@laszlodaniel 
Yes I can read wr 0031 rd 2f 
and wr 0032 rd 2f 
Step 1 and 2 good
Thanks now I will move on. What will come of this I will write. It `s Magic

bacrua, 2020-10-07:
Hi. There is no result yet. Either I am doing it wrong or the password did not fit. I am not getting access.

laszlodaniel, 2020-10-07:
Yeah, I have a problem understanding how 457 bits of input data equals 512 bytes too. But this is what every datasheet says. Perhaps there's a further zero-padding that is somehow obvious? I don't know what gives.

bacrua, 2020-10-07:
> 9.2 SHA-1 Description
The SHA-1 is known as a one-way hash function, meaning there is no known mathematical method of
computing the input given only the output. The specification of the SHA-1, as defined by FIPS 180–2,
states that the input consists of 512 bit blocks with a total input length less than 264 bits. Inputs which do
not conform to integer multiples of 512 bit blocks are padded before any block is input to the hash
function. The SHA-1 algorithm outputs 160 bits, commonly referred to as the digest.
(As of April 23, 2004 the latest revision is FIPS 180–2).SHA-1 or secure hash algorithm is used to
compute a condensed representation of a message or data also known as hash. For messages < 2^64
the SHA-1 produces an 160-bit output called digest.
The device generates an SHA-1 input block of 288 bits (total input = 160 bit message + 128 bit key). To
complete the 512 bit block size requirement of the SHA-1, the device pads the key and message with a 1,
followed by 159 0’s, followed by the 64 bit value for 288 (000¼00100100000), which conforms to the pad
requirements specified by FIPS 180–2.
Detailed information about the SHA-1 algorithm can be found:
1. http://www.itl.nist.gov/fipspubs/fip180–1.htm
2. http://csrc.nist.gov/publications/fips
3. www.faqs.org/rfcs/rfc3174.html

I understood like this:
followed by 159 0’s, followed by the 64 bit value for 288 ( 0x0120 4 repetitions ) 0x0120012001200120

bacrua, 2020-10-07:
with a 1,
followed by 159 0’s, followed by the 64 bit value for 288 (000¼00100100000), 

KD+M+80000000000000000000000000000000000000000120012001200120
or
default pass+rd 2f register+80000000000000000000000000000000000000000120012001200120

laszlodaniel, 2020-10-07:
Okay so again I can't count for my life. :'D Figures.
I took the 100100000 binary value literally and added to the last zero of the 159 zeros without padding the initial value on the left.
Still the datasheet section 9.5 says 512 bytes when in reality it's 512 bits and fails to bring up that 288 needs to be repeated 4 times. Again this would be evident for someone with cryptographic knowledge, not me.

Problem remains that the default key doesn't work for you. Have you tried it for both unseal and full access mode?

bacrua, 2020-10-07:
Been so and so. I am also not good at cryptography, I just started counting the number of bits, read the documentation several times. I can be wrong. I especially can't understand the combination -" for 288 (000¼00100100000) " 000 1\4???????????

bacrua, 2020-10-07:
https://www.liavaag.org/English/SHA-Generator/

bacrua, 2020-10-11:
There were questions. In the smshina documentation, when recording, the address of the device is transmitted only once.
https://www.ti.com/lit/an/slua475/slua475.pdf
 You have 2 times. How is it correct? Is it possible that when writing to 2F during BQ30z55 authentication, an extra or incorrect byte is written? and Because of this error?

laszlodaniel, 2020-10-11:
The seemingly double address write is normal and a requirement of the I2C-bus.

Whatever the target operation (read/write) first a start command is sent which needs to contain the device address shifted 1 bit to the left and the write-bit in the MSB position:
```
i2c_start((sb_address << 1) | I2C_WRITE);
```
Then the target register address is written:
```
i2c_write(reg);
```
Then comes another start command where the target operation is shown (either I2C_READ or I2C_WRITE). This is where the device address appears for a second time:
```
i2c_rep_start((sb_address << 1) | I2C_WRITE);
```
Then the data byte(s) need to be read or written. Word and block data needs to be separated into bytes:
```
i2c_write(data1);
i2c_write(data2);
...
i2c_write(dataN);
```
or
```
uint8_t b1 = i2c_read(false);
uint8_t b2 = i2c_read(true);
```
Notice that the i2c_read command has a boolean parameter which needs to be true only at the last read. If you only read 1 byte then it's true right away.

Finally the I2C read/write session is closed with
```
i2c_stop();
```

bacrua, 2020-10-11:
All this is clear to me. I already got into the code of Android and C#. That is why I applied. The i2c bus is just a way of transferring from device to device (physical layer). SMbus is a protocol (or application) layer, so I gave you a link to a document that describes the smbus protocol, which requires only transmitting the device address once when recording.
Theoretically, when writing, after the first transmission of the address, the receipt of data begins on the recipient's side. And this can lead to data shift.
On the other hand, the repeated start signal i2c and the transfer of the address with the WRITE bit equalizes the situation.

bacrua, 2020-10-11:
I also found disagreements in the documentation for the controller. In one chapter they write M+KD and in detailed instructions KD+M. And the description of the commands is again M+KD.
I continue to work, I will write on the results.

bacrua, 2020-10-13:
There is news. The site for receiving SHA1 translates the hexstring, but you need to translate the HEX in its raw form. Added UNSEAL access function to the C# code. While debugging and checking, I changed the factory settings. How it happened I did not understand. But the fact that it worked is there.
Conclusion: the factory password is suitable, the programs work and it is possible to gain access. An error in actions leads to loss of settings and cannot be corrected by this program. The experiment ended in failure.

laszlodaniel, 2020-10-13:
That's unfortunate. What registers were you tampering with? Or did you changed flash memory content directly? Backups?

bacrua, 2020-10-13:
I tried to access and wrote C # code. How it happened that I changed the data I do not understand. Probably got access unnoticed from me, and then wrote something down somewhere. I gave up experimenting so far.

bacrua, 2020-10-13:
I agree. Until I stop, there is not yet another test subject. I had one personal battery. I will look for an inexpensive second.

bacrua, 2020-10-13:
@laszlodaniel 
and thanks for your support.

abkarino, 2021-12-19:
@bacrua Can you share your code?
I have a battery which had 1 failing cell and I changed it and need to remove the PF flags.
Also can you explain where or why you needed to replicate 0x0120? as far as I understood, I only need to pad the 9 bits of 288 to be 64 bits i.e 0x0000000000000120.

> Task 1. Appending Padding Bits. The original message is "padded" (extended) so that its length (in bits) is congruent to 448, modulo 512. The padding rules are:
The original message is always padded with one bit "1" first.
Then zero or more bits "0" are padded to bring the length of the message up to 64 bits fewer than a multiple of 512.
Task 2. Appending Length. 64 bits are appended to the end of the padded message to indicate the length of the original message in bytes. The rules of appending length are:
The length of the original message in bytes is converted to its binary format of 64 bits. If overflow happens, only the low-order 64 bits are used.

> append ml, the original message length in bits, as a 64-bit big-endian integer.

closed 2021-12-19 by bacrua