laszlodaniel/SmartBatteryHack #14: Bq20z45

mossana, 2021-01-04:
Hello
         I have battery asus k53 . Battery have bq20z45
I connected to arduino mega 2560 but not found smbus (smart Battery hack) software
[image]
 Can you help me.
Are bq20z45 have a different connection ?

laszlodaniel, 2021-01-04:
Hi, check the wiring, try switching SDA/SCL pins around.

mossana, 2021-01-04:
I check , in  Arduino code 
[image]
Sda_pin 0
Scl_pin 1
I want change it to 
Sda_pin 20
Scl_pin 21
But give me error

mossana, 2021-01-04:
My connection
[image]

mossana, 2021-01-04:
Error when I changed pin 
[image]

laszlodaniel, 2021-01-04:
Arduino Mega labels are not the same as the port/pin definitions.
PORTD 0 -> 20 (SDA)
PORTD 1 -> 21 (SCL)

Leave the original defines as they are by default.

I see the controller board is out of the battery unit. Is it powered up properly on all of its pins?
I tried the same with a bare board earlier, connected it to 12.6V, left the innermost cell pins disconnected and the Arduino didn't see the SMBus either. Then tried to tinker around until it released the magic smoke... Be careful.

mossana, 2021-01-04:
I connected to battery 12.6
And I see it by i2c_scanner 
But when I use your software I can't connect (no  device give me )

laszlodaniel, 2021-01-04:
Same here. Something's wrong with the address scanner in this particular sketch.
It's weird because it worked before.
I'll try to figure something out.

mossana, 2021-01-04:
Ok thank you very much 💝

laszlodaniel, 2021-01-04:
Check out the update. New Arduino sketch and GUI is uploaded. The communication is still choppy.
I'm still wondering what went wrong and why the code is acting up.

mossana, 2021-01-05:
Thank you sir very much am glad for your help

laszlodaniel, 2021-01-05:
You're welcome. Keep an eye out for further updates. The code needs more fixing.
I have other projects going so if you come up with solutions open a pull request.

closed 2021-01-05 by laszlodaniel

H-Masoumi, 2025-04-07:
Hi
I getting No SMBuss Device for BQ20Z75 too.
Output pins of battery are zero (P+ and P-). But the battery cells are full charge.
Also is this project dead? There is no new developments.