laszlodaniel/SmartBatteryHack #15: Bq20z45 read info

mossana, 2021-01-05:
Hello again 😊
When I read data it appears as 
[image]
[image]

But other information not appear like 0x17 cycle count because it hex word so 
 in case 0x17:. //HEX word is..... false it must be 
Unsigned int 

[image]

mossana, 2021-01-05:
A code from smartbatteryhack......
 mainform.cs

laszlodaniel, 2021-01-06:
The GUI version is reverted to a previous version where this data is not processed yet.
Another contributor made the latest change to the GUI and it seems to be unstable.
I upload it again for you to try.

mossana, 2021-01-08:
thank you i try it now

mossana, 2021-01-08:
[image]
Hello i have two different reads of cycle count
[image]
cycle count = 928 is at true
can you see it what is the problem?

laszlodaniel, 2021-01-08:
I see if multiple registers are read then returned data gets scrambled. Not just cycle count but other registers look wrong to me.
It will take some time to fix this. I open another branch for contributions and return the master branch to a version where everything was fine.

mossana, 2021-01-08:
Ok man you are professional ☺️

mossana, 2021-01-13:
Any news ☺️ Mr

laszlodaniel, 2021-01-15:
I reverted the project to a previous version and added missing information in the GUI register dump list.
Also SMBus communication is fixed. Try it now.

mossana, 2021-01-15:
Thanks man I we'll try it 🕺

mossana, 2021-01-18:
Thanks man for your help 🤗🤗
But I can't modify a cycle count to 0 😓😓

laszlodaniel, 2021-01-18:
What's your process step-by-step?

mossana, 2021-01-18:
[image]
[image]
[image]
[image]

laszlodaniel, 2021-01-18:
You need to unseal the device first. Right now it's in sealed mode and the registers are read-only.

Read OperatingStatus (54) register first and note the value.

Then send the unseal key in two parts to ManufacturerAccess register:
Register: 00
Write word: 0414
Write word: 3672

I'm not sure this is the valid unseal key. Word order and byte order in the words may not be right. Try every variation you can.

Then read the OperatingStatus (54) register again and see if one of the bits changed. That's the sealed/unsealed status bit.

mossana, 2021-01-18:
[image]
[image]