laszlodaniel/SmartBatteryHack #24: DESIGN VOLTAGE IS 0.0 V, I tried with 3 batteries

anandakrishnas, 2022-12-19:
battery detected in GUI, but shows this result as design voltage is 0.0 V. not reading my battery. I tried 3 batteries and reads same design voltage 0.0V. Is tehre a problm with my code? am using ARDUINO UNO BOARD

[INFO] Connecting to COM12

[<-TX] Handshake request (COM12)
3D 00 02 01 00 03

[RX->] Handshake response
3D 00 08 81 00 53 42 48 41 43 4B 35

[INFO] Handshake OK: SBHACK

[INFO] Device connected (COM12)

[RX->] Device settings
3D 00 05 83 01 03 00 00 8C

[INFO] Word byte-order: reverse read/write

[INFO] Design voltage: 0.0 V

any ideas? am new to this

laszlodaniel, 2022-12-19:
That's some kind of bug. I remember getting this, just ignore it. Read registers to get accurate information!

anandakrishnas, 2022-12-20:
Yes that's fine for me, I read registers and got the values, but not able to write the cycle count. 
So I tried with many different possible unseal key to 00 address. nothing worked.
Reading from register 54 is

[INFO] Reg.: 54
       Data: 00 00

[INFO] Reg.: 2F
       Data: 31 0B

then 0x310b = 0b11000100001011
LSB =  00001011

and as per your documentation

b7 | b6 | b5 | b4 | b3 | b2 | b1 | b0
PRES | EDV2 | SS | VDQ | AFEFAIL | PF | CVOV | CVUV
0 | 0 | 0 | 0 | 1 | 0 | 1 | 1

so SS = 0 PF = 0, then no sealing i can reset cyclecount to zero. but not worked. 

Then I found this project 
https://github.com/ArminJo/Smart-Battery-Module-Info_For_Arduino
and it shows 

Pack config and status              0x310B | 0b11000100001011
                                    - Pack not inserted
                                    - Voltage > EDV2
                                    - Pack unsealed
So theoretically I can reset the cycle count to 0.

but unable to do that. can u point out the issue, below is my reg dump

[INFO] SMBus register dump details (00-50):
[00]: 00 18 // ManufacturerAccess: 00 18
[01]: 01 DB // RemainingCapacityAlarm: 475 mAh
[02]: 00 0A // RemainingTimeAlarm: 10 minutes
[03]: E0 00 // BatteryMode: 1110000000000000
[04]: 00 00 // AtRate: 0 minutes
[05]: FF FF // AtRateTimeToFull: 65535 minutes
[06]: FF FF // AtRateTimeToEmpty: 65535 minutes
[07]: 00 01 // AtRateOK: true
[08]: 0B D8 // Temperature: 27.59°C
[09]: 2A 04 // Voltage: 10.756 V
[0A]: 00 00 // Current: 0 A
[0B]: 00 00 // AverageCurrent: 0 A
[0C]: 00 01 // MaxError: 1%
[0D]: 00 24 // RelativeStateOfCharge: 36%
[0E]: 00 1F // AbsoluteStateOfCharge: 31%
[0F]: 04 E5 // RemainingCapacity: 1253 mAh
[10]: 0D 88 // FullChargeCapacity: 3464 mAh
[11]: FF FF // RunTimeToEmpty: 65535 minutes
[12]: FF FF // AverageTimeToEmpty: 65535 minutes
[13]: FF FF // AverageTimeToFull: 65535 minutes
[14]: 0A F0 // ChargingCurrent: 2.8 A
[15]: 30 0C // ChargingVoltage: 12.3 V
[16]: 00 80 // BatteryStatus: 0000000010000000
[17]: 02 9A // CycleCount: 666
[18]: 0F C8 // DesignCapacity: 4040 mAh
[19]: 2A 30 // DesignVoltage: 10.8 V
[1A]: 00 31 // SpecificationInfo: 0000000000110001
[1B]: 41 32 // ManufactureDate: 2012.09.18
[1C]: 15 EC // SerialNumber: 15 EC
[1D]: FF FF // FF FF
[1E]: FF FF // FF FF
[1F]: FF FF // FF FF
[20]: 4C 06 // ManufacturerName: 4C 06
[21]: 4C 0B // DeviceName: 4C 0B
[22]: 4C 04 // DeviceChemistry: 4C 04
[23]: 03 0E // ManufacturerData: 03 0E
[... 195 lines cut ...]
[E7]: FF FF // FF FF
[E8]: FF FF // FF FF
[E9]: FF FF // FF FF
[EA]: FF FF // FF FF
[EB]: FF FF // FF FF
[EC]: FF FF // FF FF
[ED]: FF FF // FF FF
[EE]: FF FF // FF FF
[EF]: FF FF // FF FF
[F0]: FF FF // FF FF
[F1]: FF FF // FF FF
[F2]: FF FF // FF FF
[F3]: FF FF // FF FF
[F4]: FF FF // FF FF
[F5]: FF FF // FF FF
[F6]: FF FF // FF FF
[F7]: FF FF // FF FF
[F8]: FF FF // FF FF
[F9]: FF FF // FF FF
[FA]: FF FF // FF FF
[FB]: FF FF // FF FF
[FC]: FF FF // FF FF
[FD]: FF FF // FF FF
[FE]: FF FF // FF FF
[FF]: FF FF // FF FF

laszlodaniel, 2022-12-22:
The unsealing part got me too. Never succeeded to really hack the BQ8050 so this project was never finished.
If you are dealing with this same chip then unsealing won't be easy.

Check this [wiki post](https://github.com/laszlodaniel/SmartBatteryHack/wiki/Basics) and review SHA-1 methods.
BQ8050 datasheet provides default keys to do authentication. This method is not coded into the Arduino sketch.

anandakrishnas, 2022-12-26:
yes, i got another lenovo battery.
still not able to reset cycle count, 
i will try sha 1 method

anandakrishnas, 2022-12-27:
using a code i got this as result

OperationStatus: 0
PF Status: FFFF
OperationStatus: 0
PFKey: FFFFD902
UnSealKey: FFFF3300
FullAccessKey: FFFF3300
PF Status: FFFF

laszlodaniel, 2022-12-30:
> so I must use 2F instead of 00 , right?

Yes. I have never tried this SHA-1 method but those results don't look right to me.

anandakrishnas, 2022-12-31:
> > so I must use 2F instead of 00 , right?
> 
> Yes. I have never tried this SHA-1 method but those results don't look right to me.

My knowledge level not much to do this, so am reading and watching videos. If i found anything new i will inform you. thank you for your support.

laszlodaniel, 2022-12-31:
Arduino example:
https://mega.nz/file/SB0CESqa#ZUIKriBzbMii_4iP-8wzA4IJ3pKgHWK-KrFmZAFv6-8

The key takeaway is this function:
```
// Calculate secret key from security key KD and input byte array M
void SecretKeyCalculator(uint8_t *KD, uint8_t *M) {
    // Create a SHA-1 hash object
    SHA1 sha1;

    // Generate SHA-1 input block Block1 of 512 bits
    memcpy(Block1, KD, 16);
    memcpy(Block1 + 16, M, 20);
//    Block1[36] = 0x80; // added to block by SHA1 library
//    memset(Block1 + 37, 0, 23);
//    Block1[62] = 0x01;
//    Block1[63] = 0x20; // 0x0120 = 288

    // Generate SHA-1 hash HMAC1 using Block1
    sha1.reset();
    sha1.update(Block1, 36);
    sha1.finalize(HMAC1, 20);

    // Generate SHA-1 input block Block2 of 512 bits
    memcpy(Block2, KD, 16);
    memcpy(Block2 + 16, HMAC1, 20);
//    Block2[36] = 0x80; // added to block by SHA1 library
//    memset(Block2 + 37, 0, 23);
//    Block2[62] = 0x01;
//    Block2[63] = 0x20; // 0x0120 = 288

    // Generate SHA-1 hash HMAC2 using Block2
    sha1.reset();
    sha1.update(Block2, 36);
    sha1.finalize(HMAC2, 20);
}

void setup() {
    // Example calculation using SecretKeyCalculator function
    uint8_t KD[16] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00};
    uint8_t M[20] = {0xC8, 0x2C, 0xA3, 0xCA, 0x10, 0xDE, 0xC7, 0x26, 0x8E, 0x07, 0x0A, 0x7C, 0xF0, 0xD1, 0xFE, 0x82, 0x20, 0xAA, 0xD3, 0xB8};
    SecretKeyCalculator(KD, M); // updates global byte arrays

    // Expected results:
    // HMAC1: EBF44E83 D792151C 8BE508BB 6D517C69 B331C0CE
    // HMAC2: FB8A3424 58E0B136 988CB520 3BB23F94 DFD4440E
}
```

Found an excellent Python implementation of the whole SMBus communication, including the unlock functions, both 2 word and SHA-1:
https://github.com/o-gs/dji-firmware-tools/blob/master/comm_sbs_bqctrl.py

It's basically this Arduino project but better.

anandakrishnas, 2023-01-02:
> Arduino example: https://mega.nz/file/SB0CESqa#ZUIKriBzbMii_4iP-8wzA4IJ3pKgHWK-KrFmZAFv6-8
>
> The key takeaway is this function:
>
> ```
> // Calculate secret key from security key KD and input byte array [...]

this one looks cool, i will chk today

anandakrishnas, 2023-01-06:
I CHANGED MY CYCLE COUNT TO THIS , 

[INFO] SMBus register dump details (17-17):
[17]: 12 34 // CycleCount: 4660

Previous value [17]: 02 9B // CycleCount: 667

DONT KNOW WHAT HAPPEND I TRIED WITH ANOTHER CODE TO WRITE AND READ , then i entered a random hex 1234 now showing this, will update my work

I used 51F51 chip

tomeqtl, 2023-06-01:
i cant connect smart battery hack with my 51f51 battery, adruino uno
`[INFO] Connecting to COM4

[<-TX] Handshake request (COM4)
3D 00 02 01 00 03

[RX->] Data received
06 18 78 F8 86 E0 98 E0 18 66 00 18

[INFO] Handshake ERROR: ???f 

[INFO] Device not found at COM4

[INFO] Connecting to COM4

[<-TX] Handshake request (COM4)
3D 00 02 01 00 03

[RX->] Data received
06 18 78 F8 86 60 98 E0 18 66 00 18

[INFO] Handshake ERROR: `??f 

[INFO] Device not found at COM4

[INFO] Connecting to COM4

[<-TX] Handshake request (COM4)
3D 00 02 01 00 03

[RX->] Data received
06 18 78 F8 86 60 98 E0 18 66 00 18

[INFO] Handshake ERROR: `??f 

[INFO] Device not found at COM4

[INFO] Connecting to COM4

[<-TX] Handshake request (COM4)
3D 00 02 01 00 03

[RX->] Data received
06 18 78 F8 86 60 98 E0 18 66 00 18

[INFO] Handshake ERROR: `??f 

[INFO] Device not found at COM4

[INFO] Connecting to COM4

[<-TX] Handshake request (COM4)
3D 00 02 01 00 03

[RX->] Data received
06 18 78 F8 86 60 98 E0 18 66 00 18

[INFO] Handshake ERROR: `??f 

[INFO] Device not found at COM4

[INFO] Connecting to COM4

[<-TX] Handshake request (COM4)
3D 00 02 01 00 03

[RX->] Data received
06 18 78 F8 86 E0 98 E0 18 66 00 18

[INFO] Handshake ERROR: ???f 

[INFO] Device not found at COM4

[INFO] Connecting to COM4

[<-TX] Handshake request (COM4)
3D 00 02 01 00 03

[RX->] Data received
06 18 78 F8 86 60 98 E0 18 66 00 18

[INFO] Handshake ERROR: `??f 

[INFO] Device not found at COM4

[INFO] Connecting to COM4

[<-TX] Handshake request (COM4)
3D 00 02 01 00 03

[RX->] Data received
06 18 78 F8 86 60 18 18 66 00 18 F8

[INFO] Handshake ERROR: `f 

[INFO] Device not found at COM4

[INFO] Connecting to COM4

[<-TX] Handshake request (COM4)
3D 00 02 01 00 03

[RX->] Data received
06 18 78 F8 86 60 18 E0 18 66 00 18

[INFO] Handshake ERROR: `?f 

[INFO] Device not found at COM4

[INFO] Connecting to COM4

[<-TX] Handshake request (COM4)
3D 00 02 01 00 03

[RX->] Data received
06 18 78 F8 86 60 98 E0 18 66 00 18

[INFO] Handshake ERROR: `??f 

[INFO] Device not found at COM4

[INFO] Connecting to COM4

[<-TX] Handshake request (COM4)
3D 00 02 01 00 03

[RX->] Data received
06 18 78 F8 86 60 18 18 66 00 18 F8

[INFO] Handshake ERROR: `f 

[INFO] Device not found at COM4

[INFO] Connecting to COM4

[<-TX] Handshake request (COM4)
3D 00 02 01 00 03

[RX->] Data received
06 18 78 F8 86 60 18 18 66 00 18 F8

[INFO] Handshake ERROR: `f 

[INFO] Device not found at COM4

[INFO] Connecting to COM4

[<-TX] Handshake request (COM4)
3D 00 02 01 00 03

[RX->] Data received
06 18 78 F8 86 60 98 E0 18 66 00 18

[INFO] Handshake ERROR: `??f 

[INFO] Device not found at COM4

[INFO] Connecting to COM4

[<-TX] Handshake request (COM4)
3D 00 02 01 00 03

[RX->] Data received
06 18 F8 F8 86 60 18 18 66 00 18 F8

[INFO] Handshake ERROR: `f 

[INFO] Device not found at COM4

[INFO] Connecting to COM4

[<-TX] Handshake request (COM4)
3D 00 02 01 00 03

[RX->] Data received
06 18 78 F8 86 60 18 E0 18 66 00 18

[INFO] Handshake ERROR: `?f 

[INFO] Device not found at `COM4``

Arduino read ok

`PF Status: FFFF
 24.15°C
atus: 0
PFKey: FFFF6201
UnSealKey: FFFFEC03
FullAccessKey: FFFF8E00
PF Status: FFFF
Voltage Temperature
12.41V  24.15°C`

closed 2023-06-04 by anandakrishnas