laszlodaniel/SmartBatteryHack #26: what is the unseal key for "045A20 SANYO chip"?

akib360, 2023-10-19:


Jlozde, 2024-04-30:
That is a nice question that I am searching. There are tools which can unlock so it is known for some parties but it is not publicly shared. Somebody needs to sniff while paid tools are unlocking. Any volunteer :)

GZQUICK, 2024-10-26:
it is renesas chip

Vss1272, 2025-03-17:
[image]

heres screen from paid software, and dump of eeprom, somewhere theree is pass

Pikolodaimaku, 2025-09-07:
> [image]
> 
> heres screen from paid software, and dump of eeprom, somewhere theree is pass

If you share the dump we can get the key but with just a capture it is impossible

zoki2334, 2025-11-02:
@Vss1272
Did you find the unseal key? I have the same problem and I'm looking for the unseal key too.