laszlodaniel/SmartBatteryHack #27: Can't read 160 bit message M to set FullAccess (BQ30Z55)

JhonnyAndreatta, 2024-03-26:
Hello, I'm trying to enter on the full access mode for BQ30Z55 chip, but this model uses the SHA-1 encryption, and I need to retrieve the 160 bit key to unlock the chip. However, when I follow the steps present on the wiki, when trying to read the 160 bit key after sending 0x0032 to ManufacturerAccess (00), the response that I got on the terminal is something like: ????????????????????, or something like that but with some letters in the middle. I'll put a picture on this thread to better explanation. 
[image]
I'm currently using Arduino MEGA (Atmega 2560 chip), hosted by an CH340 USB-Serial Chip.
Someone who knows what I'm doing wrong?
Thanks!!

JhonnyAndreatta, 2024-03-26:
Oh I get it now, I wasn't understanding the block line correctly, thanks!
I'll try to proceed with the unlocking...
Now another question, the key can be in hex or I need to convert the hex string to another base, like decimal or Base64? 
And to generate the B1 block, where can I get it the 128-bit unseal/full access key KD? I'm kinda confused about this part...
Thanks a lot! Have a nice day.

laszlodaniel, 2024-03-27:
Default unseal/full access key for the BQ30Z55 chip is:

```
0x0123456789ABCDEFFEDCBA9876543210
```

No need to convert this number to another base, just make sure to follow the guide.

There was an issue opened about this chip a few years back:
https://github.com/laszlodaniel/SmartBatteryHack/issues/13

Good luck!

JhonnyAndreatta, 2024-03-29:
Thanks for the default key!! That earned me a lot of time!
Reading the other thread I see that the author of the thread didn't pass the calculation of SHA-1 hash.
Researching about HMAC SHA-1, and reading the datasheet itself, I founded that the hash is generated using a public message and a private key. But on the datasheet, Texas doesn't tell us what's the secret key to use on the HMAC system. 
I'm little bit confused, I'll try to use the public message being the block B1, and the private key the default 128-bit key that you provided on the upper post. 
Texas doesn't seems to care about the final consumer, their datasheet is a lot confusing haha. 
If I get any update on this I'll post down here.
Thanks!

JhonnyAndreatta, 2024-04-02:
Hello.
So to solve the problem to unseal the BQ30Z55, today I bought the CP2112 board, that's a dev board which performs a SMBus to USB conversion, with that, I was able to use the DJI Battery killer, which is a program to reset the chips on drones batterys. The program is able to unlock and guarantee full access to the chip, he does the cryptography calculations automatically, and unlocks it. 
In the future I'll try to read the calculations provided by the software for better understanding of the description process. If I get somewhere I'll create a new thread here to explain how to unseal the BQ30Z55 with the Arduino.
Thanks a lot for your help and your time!!!

closed 2024-04-02 by laszlodaniel